SMPTE/ris-osvp-metadata-camdkit #17: Consider refactoring sensor_pixel_dimensions and sensor_physical_dimensions with sampled_pixel_pitch

palemieux, 2022-05-04:
It is not clear how `sensor_pixel_dimensions` and `sensor_physical_dimensions` are useful to OSVP:

- OSVP receives the sampled/active image area, which does not always correspond to the sensor dimensions
- isn't the critical quantity the pixel pitch for OSVP?
- would OSVP use sensor_pixel_dimensions and sensor_physical_dimensions for any other than to compute pixel pitch?
- the pixel pitch is independent of cropping (it is dependent on scaling)

palemieux, 2022-05-05:
# 2022-05-05 meeting

`sensor_physical_dimensions` are used for visual effects (would be good to have the kind of computation it is used for)

repentsinner, 2022-05-05:
There was a note in the Zoom chat about it being used to compute the frustum for somewhat more accurate sizing/rendering to the LED.

palemieux, 2022-05-05:
@repentsinner How does this computation work? Also, it is the actual sensor size, or the size that was used to sample the image?

repentsinner, 2022-05-10:
I haven't done the computation myself, but we're trying to figure out the projection of the viewing cone from the camera into the world, and then calculate the intersection of that cone with the LED displays (or whatever else). In this use, it should be the size (and I guess location) that was used to sample the image for any given shot. This mechanism would also be used to simulate framing of a real sensor/lens combo during previz.

I am not sure if there are other uses that would like to know the largest size possible for a given sensor as well.

palemieux, 2022-07-10:
It looks like the physical dimensions of the image, i.e. the dimensions of the active part of the sensor, are used to calculate the angular field-of-view:

`AFOV = 2 atan(dim/2/f)`

So I suggest merely renaming `sensor_physical_dimensions` and `sensor_pixel_dimensions` to `image_physical_dimensions` and `image_pixel_dimensions` to emphasize that both measurements are for the active portion of the sensor.

palemieux, 2022-07-11:
> Per your original question: knowing pixel pitch as well as fill factor (and I suppose angular/spatial orientation) are likely useful for moire calculations,

Would that be used to match VFX graphics to the recorded image?

palemieux, 2022-07-11:
@repentsinner Is there literature available on the process of estimating Moire based on the sensor characteristics? We should make sure to capture the right information, e.g. photosite layout, etc.

repentsinner, 2022-07-11:
I'm not aware of any specific to VP, it's more of a general sampling theorem topic that I'm not certain has been applied with much rigor to VP system design yet. It is on my (long) list of topics to explore and understand.

I guess there are two layers to it - luminance moire where it doesn't likely matter so much the photosite filter (RGB or whatever, and subsequent bayer pattern) and then chroma moire which is also annoying but possibly less problematic which would most likely require an understanding of the photosite layout.

I've been running under the assumption that we need to double (or more) the pixel resolution of the display relative to the sensor over a given area per [Nyquist/Shannon](https://en.wikipedia.org/wiki/Nyquist%E2%80%93Shannon_sampling_theorem), however fill ratio plays a large role (e.g., a display showing flat white is actually a grid of extremely high contrast point light sources embedded in a black field that integrates to white(ish) if your sensor/eye is sufficiently outresolved (or overblown), however at an arbitrary distance it may actually moire unlike a lit white card that is painted and uniformly white at least at a macroscopic level). With this understanding, increasing fill factor may yield as much or more improvement w/r/t moire as just shrinking the display pixel pitch.

PR #21 (2022-08-11): Clarify sensor_*_dimensions
Closes #17
closed 2022-08-11 by palemieux