SMPTE/ris-osvp-metadata-camdkit #33: Add support for tint value

palemieux, 2022-09-01:


umathurred, 2022-11-03:
Uday Mathur
  [1 hour ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667504029698299?thread_ts=1667503868.165809&cid=C03EAS655DJ)
Another option would be to represent the white balance using CIE XY ... that is unambiguous

New

Tucker Downs
:cactus:  [1 hour ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667504399039119?thread_ts=1667503868.165809&cid=C03EAS655DJ)
 CIE xy (little x little y) is the same as kelvin + duv. They can be mathematically transformed. Though using x,y or u,v directly does allow for much more extreme “white points”

Uday Mathur
  [1 hour ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667504456815269?thread_ts=1667503868.165809&cid=C03EAS655DJ)
Yeah, I'm glad you mentioned that. White points for creative intent may be non-white

Uday Mathur
  [1 hour ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667504492625209?thread_ts=1667503868.165809&cid=C03EAS655DJ)
We like CIE xy because it's the same but unambiguous. Kelvin and Tint are subject to how manufacturers handle tint

Uday Mathur
  [1 hour ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667504745176459?thread_ts=1667503868.165809&cid=C03EAS655DJ)
In our conversation internally we discussed how we have changed the handling of tint over the years and how the SDK abstracts the complexity. So our proposal is to instead represent this as CIE XY . 
[@Stacey Spears](https://smpte-ris-osvp.slack.com/team/U040G6MGPMW)
, please correct me if I muddled the description

Tucker Downs
:cactus:  [1 hour ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667504801947509?thread_ts=1667503868.165809&cid=C03EAS655DJ)
Tint to me means duv which is an unambiguous definition. Non compliant manufacturers might use ambiguous methods.

Tucker Downs
:cactus:  [1 hour ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667504854616199?thread_ts=1667503868.165809&cid=C03EAS655DJ)
Sorry to be short, I’m in another meeting but this is a very important conversation!!

Uday Mathur
  [1 hour ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667504962753569?thread_ts=1667503868.165809&cid=C03EAS655DJ)
maybe we're one of those problematic non-compliant manufacturers who have changed the locus

Tucker Downs
:cactus:  [1 hour ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667505086730899?thread_ts=1667503868.165809&cid=C03EAS655DJ)
It might be very justified for engineering or artistic reasons!! But whatever the internal selection of white point follows, it can be transformed into plankian K + duv

Tucker Downs
:cactus:  [1 hour ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667505095018879?thread_ts=1667503868.165809&cid=C03EAS655DJ)
I think?

Uday Mathur
  [1 hour ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667505128067999?thread_ts=1667503868.165809&cid=C03EAS655DJ)
Yes, it may not match what we show in the camera UI or other NLEs but if we choose for compatibility to use duv and planckian, we can transform our metadata to match that

Uday Mathur
  [1 hour ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667505141415119?thread_ts=1667503868.165809&cid=C03EAS655DJ)
or just report CIE xy

Tucker Downs
:cactus:  [1 hour ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667505142119809?thread_ts=1667503868.165809&cid=C03EAS655DJ)
K + duv is desirable because it closely aligns with controls. But if it will lead to incompatibility we should avoid it.

Uday Mathur
  [1 hour ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667505282955739?thread_ts=1667503868.165809&cid=C03EAS655DJ)
I don't think it's incompatibility, just may be inconsistent with the UI reported values if we use planckian locus instead of our mixed planckian/daylight

Uday Mathur
  [1 hour ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667505332351879?thread_ts=1667503868.165809&cid=C03EAS655DJ)
I'll let 
[@Stacey Spears](https://smpte-ris-osvp.slack.com/team/U040G6MGPMW)
 speak more for Red. He can speak in more detail than I can

Tucker Downs
:cactus:  [1 hour ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667505365684459?thread_ts=1667503868.165809&cid=C03EAS655DJ)
For example, I imagine if a user selects a 5500K WB you might make the reasonable assumption that they are using natural daylight. Which has a plankian CCT of 5500K + 0.0100duv. If they then add apply a -green of 1 click, maybe the UI says that it’s 5500K - 1G but the metadata might report 5500K + 0.006duv (implying green steps of .004 at this setting)

[... 73 lines cut ...]
Tucker Downs
:cactus:  [14 minutes ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667507452103989?thread_ts=1667503868.165809&cid=C03EAS655DJ)
Grrr. Going into another meeting.

Joseph Goldstone
  [14 minutes ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667507468031319?thread_ts=1667503868.165809&cid=C03EAS655DJ)
[misery loves company]

Tucker Downs
:cactus:  [14 minutes ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667507487063329?thread_ts=1667503868.165809&cid=C03EAS655DJ)
Please see my above example where I point out that this metadata does not need to strictly match the user interface.

Tucker Downs
:cactus:  [12 minutes ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667507586154719?thread_ts=1667503868.165809&cid=C03EAS655DJ)
Re: xy and cinematographers. I completely agree which is why I support CCT duv

Uday Mathur
  [7 minutes ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667507871446599?thread_ts=1667503868.165809&cid=C03EAS655DJ)
I'm good with it too. only caveat would be that the tint value may not reflect what was displayed in camera UI for Red
:+1:
1

Uday Mathur
  [7 minutes ago](https://smpte-ris-osvp.slack.com/archives/C03EAS655DJ/p1667507881378759?thread_ts=1667503868.165809&cid=C03EAS655DJ)
but the color will be accurate

umathurred, 2022-11-03:
Tint will refer to duv and will be represented as x10000 with positive values to green and negative values to magenta
a signed 16-bit value can encode the tint

umathurred, 2022-11-03:
Joseph Goldstone referenced RDD 55 floating point representation as an alternative to fixed point that I suggested. We should discuss and choose. I'm open to either

palemieux, 2022-11-03:
@umathurred How would you extract the _duv_ value from the camera files:

https://github.com/sandflow/camdkit/blob/main/src/test/resources/arri
https://github.com/sandflow/camdkit/tree/main/src/test/resources/red
https://github.com/sandflow/camdkit/tree/main/src/test/resources/venice

umathurred, 2022-11-04:
Red currently uses a different representation for tint. We will add a sdk method to transform our internal representation to this vendor neutral representation.

palemieux, 2022-11-04:
By the way, it looks like _tint_ (and _white balance_) is not an intrinsic characteristic of the camera system but instead selectively applied to some of the outputs. It is therefore not clear to me it is an essential parameter, since it is not always used.