SMPTE/ris-osvp-metadata-camdkit #61: Add support for CameraSerialNumber

richardssam, 2023-01-26:
Camera Serial Number is part of the SMPTE 2065-4 spec, and does exist in Arri (as CameraSerialNumber) and Red (Camera_PIN), and I suspect for Sony Venice, its mxf/camera Attributes.

This is super useful for uniquely identifying the camera body, and its common to do lens charts recording both the camera body SN and the Lens SN, so we map cases where the camera body is producing an off center lens axis.

palemieux, 2023-01-26:
See https://smpte.slack.com/archives/C04E88H4FNJ/p1674757096673189 re: Sony Venice

makkbe, 2023-01-28:
I would recommend we add additional fields for camera manufacturer, model, serial number, and firmware version. These are all fields that exists in ANSI E1.59, a standard with which we should aim to be compatible.

repentsinner, 2023-01-28:
Agreed to call out those four fields as individual records to make filtering and switching easier, and hope that the vendors use semantic versioning for their firmware version info.

palemieux, 2023-01-31:
Below are the camera and lens identification fields:

* lens_make
* lens_model
* lens_serial_number
* lens_firmware
* camera_make
* camera_model
* camera_serial_number
* camera_firmware

palemieux, 2023-01-31:
@repentsinner @makkbe @richardssam Please review #62

repentsinner, 2023-01-31:
LGTM.

RED exposes user-editable SLATE_CAMERA_ID and SLATE_CAMERA_POS fields that are useful for identification, but not sure if they have parity fields from the other vendors.

PR #62 (2023-02-02): Add support for camera and lens identification information
Closes #61
closed 2023-02-02 by palemieux