SMPTE/ris-osvp-metadata-camdkit #64: Add support for user-editable camera identifiers

repentsinner, 2023-02-02:
Similar to #61, but for user-editable camera identification fields in addition to (typically) intrinsic fields. #61 is actually titled "Add support for camera and lens identification information" and doesn't specify that this request is limited to intrinsic fields, but we can split this out as a separate issue if desired.

RED exposes user-editable `SLATE_CAMERA_ID` and `SLATE_CAMERA_POS` fields that are useful for identification, but not sure if they have parity fields from the other vendors. I think Joseph mentioned that Arri has an `index` field?

richardssam, 2023-02-04:
In SMPTE 2065 this is cameraLabel which is defined as "The cameraLabel attribute shall be of type string, and shall contain a text label for how the camera is used or assigned in this clip, for example "Camera 1 Left", "B Camera", or "POV". The string shall contain only printable characters. The string shall contain at least one byte."

This matches what ARRI uses too.

JGoldstone, 2024-12-09:
`camdkit` defines the `camera_label` attribute of a `Clip` to be

> """Non-blank string containing user-determined camera identifier"""

@repentsinner , can we close this issue now?

repentsinner, 2024-12-16:
Hi @JGoldstone that field seems fine, thanks for locating it!

closed 2024-12-16 by repentsinner