SMPTE/ris-osvp-metadata-camdkit #69: Represent numerator/denominator as integers instead of strings

umathurred, 2023-02-16:
Fractions are currently represented as 
      "type": "string",
      "regex": "[0-9]{1,10}/[0-9]{1,10}"

Recommend representing as separate numerator and denominator integers

palemieux, 2023-02-17:
A significant drawback of using objects to represent rational values is that it makes arrays a lot wordier.

repentsinner, 2023-03-02:
My preference is for correctness rather than conciseness in code. We should not need to use a lot of them as statics, and can always provide transformers to/from strings in actual implementations as necessary.

makkbe, 2023-03-16:
In my opinion, option 1 e.g. `[{"num": 1, "denom": 2}, {"num": 13, "denom": 7}]` would be preferable from a parsing perspective.

PatrickPalmer, 2023-03-16:
For reference, Unreal Engine stores and uses Frame Rate as two 32-bit integers. 
 * https://docs.unrealengine.com/5.0/en-US/API/Runtime/Core/Misc/FFrameRate/

PatrickPalmer, 2023-03-16:
My personal preference is to store frame rate as two integers in an object.  For example: { "numerator" 24, "denominator": 1} for 24 fps or { "numerator" 24000, "denominator": 1001} for 23.98

repentsinner, 2023-03-16:
if we're concerned about json efficiency, we could use an array instead of an object: `[24,1], [24000,1000]`.

umathurred, 2023-03-16:
I'd like to include Joseph's comment from 68 that SMPTE ST 2065-4 also specifies how to encode rational numbers. I have never seen frame rate specified as a string, but my exposure may be narrower than other members of this group, and I'd be interested to see other places where the string representation is used.

Some video codecs avoid this question of fractions by representing frame rate as microseconds per frame.

palemieux, 2023-03-17:
To be clear, the model represents rational numbers as pair of integers in the range (0..2,147,483,647]. 

The discussion here is solely in the context of the JSON plaintext representation, which is intended to be human readable.

There will be other representations.

richardssam, 2023-03-20:
My preference is for something like {"num": 1, "denom": 2} and following unreal's example make it always be that, so if you want 30FPS it would be {'num': 30, 'denom': 1} that way we dont need to do special cases.

closed 2023-03-20 by palemieux