SMPTE/ris-osvp-metadata-camdkit #77: Cooke defines three types of focal length. Do we need to specify which we mean?

j-helman, 2023-03-02:
Nominal focal length is the number printed on the side of a prime lens, e.g. 50 mm. It’s used to unambiguously describe which lens is needed for a shot, and gives a good (but approximate) indication of the visual effect the lens will produce. This number is the one used by directors, cinematographers, camera operators, and lens rental houses.

Pinhole focal length (PFL) is used for computer generated imagery (CGI), and more generally throughout the visual effects (VFX) software pipeline. Object modelling software, renderers and matchmoving applications are all based around pinhole camera models, and hence specify lens characteristics in terms of the PFL.
When a VFX or matchmove artist talks about focal length, they’re probably talking about PFL.

Effective focal length (EFL) is part of a simplified optical model (the ‘thick lens model’). This model is important when designing lenses, and understanding certain imaging characteristics (particularly depth of field). When a lens designer talks about focal length, they’re probably talking about EFL.

j-helman, 2023-03-02:
Per group discussion, the current metadata item should be called Nominal.

j-helman, 2023-03-02:
Per group discussion, add Effective. It is available in Cooke/i S7 and S8.... Discussion of whether Pinhole should be added. Can be derived. Defer Pinhole.

palemieux, 2023-03-12:
Looking back at the discussion, I do think that `focalLength` should be renamed `nominalFocalLength` since I have not seen that practice elsewhere. Instead, I suggest stating in the description of `focalLength` that it is the nominal focal length, i.e. the one written on the side of the lens.

Other focal lengths should have a qualifier, e.g. `effectiveFocalLength`.

closed 2023-03-16 by palemieux