SMPTE/ris-osvp-metadata-camdkit #90: Change denominator check in strictly positive rational parameters

JGoldstone, 2023-05-19:
At the moment, StrictlyPositiveRationalParameter instances have a validate method that checks to see if the denominator is in the range `(0, (1<<31)-1]`. If the denominator were changed to support integers in the range `(0, (1<<31)-1]`, then strictly positive rational parameter instances could be mapped directly into OpenEXR `Rational` instances.

palemieux, 2023-07-27:
> If the denominator were changed to support integers in the range (0, (1<<31)-1]

You mean in the range _[0, (1<<31)-1]_, i.e. including 0, right?

If so, that would not be strictly positive but simply positive.

What about creating a new type called `PositiveRationalParameter`?

JGoldstone, 2023-08-03:
I think [0, (1<<31)-1] would be termed _strictly non-negative_. 
https://en.wikipedia.org/wiki/Strictly_positive_measure
https://proofwiki.org/wiki/Definition:Strictly_Positive/Integer

And there aren't a lot of cases in `camdkit` where I would legitimately want a zero denominator.

palemieux, 2023-09-13:
We should also fix the typo `StrictlyPositveRationalParameter` -> `StrictlyPositiveRationalParameter`

palemieux, 2023-09-21:
@JGoldstone Can you confirm that your initial request was to change the range of the denominator to `(0, (1<<32)-1]`?

palemieux, 2023-09-27:
Replaced by #98

closed 2023-09-27 by palemieux