SMPTE/ris-osvp-metadata-camdkit #92: Use Black to make camdkit PEP8-compliant

JGoldstone, 2023-05-19:
IDEs like PyCharm that inspect for PEP8 compliance can detect edge cases worth addressing, but the genuinely concerning issues can be buried in copious output if the code is far away from being PEP8-compliant.

Maintaining code for PEP8 compliance is tedious but can be automated. A tool for such automation is Black:
[https://github.com/psf/black](url)

For those of you who work in Python, would you be willing to give yourselves over on matters of style to Black?

palemieux, 2023-07-27:
The codebase is already linted using pylint. What would be the advantage of switching?

JGoldstone, 2023-09-22:
None. I had not realized it was already going through pylint. Please close this issue (or if you prefer, tell me to do so).