SMPTE/ris-osvp-metadata-camdkit #93: Scope of uniqueness of cameraSerialNumber and lensSerialNumber needs clarification

JGoldstone, 2023-05-19:
Currently for both `cameraSerialNumber` and `lensSerialNumber`, the docstrings simply say "Unique identifier of the camera" or "Unique identifier of the lens". This could be interpreted as saying cameraSerialNumber, e.g., needs to be globally unique. In fact, the cameraSerialNumber only needs to be unique in the context of the camera make and camera model, and the lens serial number only needs to be unique in the context of the lens make and lens model.

palemieux, 2023-07-27:
>  In fact, the cameraSerialNumber only needs to be unique in the context of the camera make and camera model, and the lens serial number only needs to be unique in the context of the lens make and lens model.

I think these would be good clarifications.

JGoldstone, 2023-08-03:
This is now pull request #94

closed 2024-10-10 by jamesmosys