SMPTE/ris-osvp-metadata-camdkit #107: protocol and protocol version: making them less generic, and expressing a dependency

JGoldstone, 2024-10-06:
As the project is still camdkit, and not OpenTrackIOKit, one can imagine others building a different application with new application-specific metadata of their own added on top of core camdkit metadata perhaps used in OpenTrackIO, perhaps not.

The comment for ProtocolVersion in model.py reads "Free string that describes the version of the OpenTrackIO protocol that this sample employs." Through what was probably a copy-and-paste error, the same string is used to describe the Protocol property too.

I suggest two things:

1. Rename protocol to trackingProtocol and protocolVersion to trackingProtocolVersion. That way if, say, someone re-used components of camdkit in a new protocol for monitoring which physical assets were in use at any one time for fine-grained client accounting, we would have 'trackingProtocol' and 'assetMonitoringProtocol' instead of the more awkward 'protocol' and 'assetMonitoringProtocol', the latter sounding like a special case of the more general former.
2. Change the comment for TrackingProtocol to be "Free string that describes the tracking protocol used to create this sample" and change the comment for TrackingProtocolVersion to be "Free string that describes the version of the tracking protocol, named in TrackingProtocol, used to create this sample".

That said, one issue raised here is that the presence of trackingProtocolVersion really necessitates the presence of trackingProtocol to be useful, and we don't have a way to express that one property requires another. A way around this would be to change trackingProtocol from being a free string to being a tuple of *two* free strings, the first being the name of the tracking protocol and the second the version of the tracking protocol.

jamesmosys, 2024-10-07:
What about keeping protocol and protocolVersion but updating the description to make it more generic? E.g. "In the case of a protocol, this is a free string by which it is uniquely identified" and "In the case of a protocol, this is a free string which uniquely identifies its version".
To the second point I think for readability and parseability it is easier to keep them separate, but I am not set on this!

repentsinner, 2024-10-08:
Would a dictionary be better (more self-descriptive) than a tuple?

Also if possible can we put a constrain that the version string must follow [semantic versioning](https://semver.org/)?

```javascript
protocol = {
  name: "OpenTrackIO",
  version: "0.1.0"
}

protocol = [ "OpenTrackIO", "0.1.0" ]
```

JGoldstone, 2024-10-09:
Right now the PR'd code checks that all three components are numeric, but doesn't go the whole distance with semantic versioning, which would allow hyphen-offset special designators and build markers:

[image]

It does so with this:

[image]

Shall we say that's enough for now?

More importantly, if you want to follow semantic versioning (and I favor that idea) then we need to decide if what we will expose at the time of the MTC is a 0.x.y release or a 1.0.0 release. Semantic versioning heavily constrains what we can do if it's a 1.0.0.

My personal feeling is that though we are nominally doing our work as a public CD, by holding off on merging from one of a near-dozen branches until we think we have 1.0, we are weakening the benefits of being public; I would like us to put something on the order of a 0.8.y tag on main after we merge in juren-trackerkit.

JGoldstone, 2024-10-09:
For reference:
[image]