SMPTE/ris-osvp-metadata-camdkit #127: Some Tuple type-hints should be variadic or more specific (or not there at all)

JGoldstone, 2024-12-09:
The type hints in the definition of `Distortion` are currently

> `radial: typing.Tuple[float]`
> `tangential: typing.Optional[typing.Tuple[float]] = None`
> `model: typing.Optional[typing.Tuple[str]`

An example `Distortion` from `test_model.py` is `Distortion([1.0,2.0,3.0], [1.0,2.0], "Brown-Conrady D-U")`. Some observations:
- the type hint for `Distortion.radial` only allows a single-element `tuple` but the example above provides three elements. If we believe (I don't) that all radial distortion models have three elements then the proper hint is `Tuple[float, float, float]` or if you know you are running Python 3.9 or later, `tuple[float, float, float]`. More flexibly, if we want to allow any number of floats for radial distortion we can type-hint as `typing.Tuple[float, ...`] or `tuple[float, ...]`. This latter style (whether from `typing` or `builtin`) is the one I would probably recommend.
- the type hint for `Distortion.tangential` has the same issue.
- the use of `typing.Tuple` in the definition of the `Distortion.model` attribute seems wrong.

The type hint for `VersionedProtocol.version` should be `typing.Tuple[int, int, int]`.

My IDE (PyCharm) flags the above issues; it doesn't flag the arguments to the assignments of clip attributes in (_e.g._) `test_serialize()` in `test_model.py`, probably because when the descriptors implementing the clip attributes are created, they have no type hints.

If there's consensus that we know the type of arguments above but not their number (except for the case of `VersionedProtocol.version`) I would be happy to make those hints variadic.

h-fehlis, 2024-12-11:
The number of radial distortion coefficients is anything between 1 and 6 (currently) but there is a (very slim) chance of needing more in the future, so that should be a variadic structure allowing any number of elements. The same is true for the tangential coefficients, being a maximum of 2 currently but in the future there is a (very slim) chance of needing more. But even the protocol version could have anything between 1 and 3 elements (and even more? hopefully not), so [1] means version 1.0.0, [1, 2] means version 1.2.0, [1, 2, 3, 4] would be version 1.2.3.4 (I don't think we should go that far, though, and stop at 3 elements).

JGoldstone, 2024-12-11:
I've actually seen a fourth component on versions but it's usually not numeric; it's some sort of alpha, beta, or other "hey this isn't even fully baked for a patch release" designator.

Going back to distortion models and distortion coefficients though, as @jamesmosys elsewhere pointed out (or perhaps implied by something he pointed out) there are interesting implications for what `Clip.append()` should do if incoming samples have multiple sets of distortion coefficients because a device is sending simultaneously more than one model's worth, e.g. a device might package together distortion coefficients for Brown-Conrady, 3DE and an in-house (think Wētā, ILM, Framestore-level facility) model.

Even if one sets aside multiple simultaneous models' coefficients arriving, should `Clip.append()` insist that the model stay the same when appending, and that number of coefficients for each component (radial and tangential) likewise remains constant?

jamesmosys, 2024-12-12:
In general the JSON is extensible, so anyone can add whatever they like - other distortion models etc. Since in this case any such additional lens model would need to be implemented on both the producer and the consumer, it can just be agreed between them and added. I maintain that we shouldn't 'officially' support more than one model... for now at least! (I'll ref this response in the other issue too.)

On the typing specification I will fix now in my current PR to make the fields you point out more specific...

jamesmosys, 2024-12-13:
I like that - great idea - we can even make the 'model' key optional (defaulting to Brown-Conrady D-U), so a for a common simple packet we have just introduced two characters '[]' as so:
```
"lens": {
  "distortion": [
    {
      "radial": { ... }
      "tangential": { ... }
    }
  ]
}
```
Give this a thumbs up and I'll implement in the current open PR. I will add the more complex example you give too.

JGoldstone, 2024-12-13:
Since we now have the ability to inject previously unknown distortion and undistortion model into the JSON without needing to revise the schema, can we now get rid of the 'custom' lens parameter?

jamesmosys, 2024-12-14:
Agreed. Will remove.

jamesmosys, 2024-12-17:
Resolved in https://github.com/SMPTE/ris-osvp-metadata-camdkit/pull/125