SMPTE/ris-osvp-metadata-camdkit #128: Range limits on Rotator3

JGoldstone, 2024-12-09:
If one looks at [the Wikipedia article on Euler Angles](https://en.wikipedia.org/wiki/Euler_angles), in the section on Tait-Bryan angles, the closes example one finds to pan, tilt and roll are the heading, elevation and bank angles for aircraft.

We don't currently check angle ranges in `Transform`'s use of Rotator3. Should we adopt the convention that pan and roll are inclusively within [0.0º, 360.0º] and tilt is inclusively within [-90.0º, 90º] ?

h-fehlis, 2024-12-10:
I would like to propose limits [-180, +180] for both pan and roll (instead of [0, 360]) because the camera's pan and roll angle will often be around 0 and thus we avoid a frequent "wrap around". The wrap around will create problems if somebody interpolates angles without taking care of these potential jumps in the data.
Also, in some situations it might be an advantage to be able to use a pan angle of, for example, 540 degrees to signal that the camera rotated by one and a half full revolutions to the left since tracking started (rather than just half a revolution). This might tell an operator to watch out for problems with camera cables being wound up too much. Similarly, a tilt angle of -95 degrees could tell an operator that the camera was tilted down all the way and then 5 degrees more, so in order to return to zero it should be tilted up by 95 degrees rather the tilted up by 85 degrees (180 - 95) and rolled by 180 degrees (which it was not) and panned by 180 degrees (which it also was not), both roll and pan being a result of handling the wrap around. This has nothing to do with the image of the virtual camera (it is the same whether we use wrap arounds or not) but in my experience operators sometimes look at the tracking data to see how far the camera has moved.
TL;DR: So there might be reasons not to enforce limits on pan, tilt and roll at all.

jamesmosys, 2024-12-12:
I agree that we should not enforce limits. One of the reasons to use Euler angles is so that multiple rotations can be encoded (not possible with a quaternion representation).

jamesmosys, 2024-12-17:
@JGoldstone can this be closed?

closed 2025-01-06 by jamesmosys