SMPTE/ris-osvp-metadata-camdkit #158: Move `distortionOverscan` and `undistortionOverscan` into distortion array elements

JGoldstone, 2025-02-08:
Quoting Hendrik:

> In the "lens" object there are entries `distortionOverscan` and `undistortionOverscan` which, I believe, should be in the respective objects in the `distortion` array because if we ever add more lens models then it is not clear which model these entries refer to.
> So instead of:

```python
"lens": {
  "distortionOverscan": 1.1,
  "distortion": [
    {
      "model": "Brown-Conrady D-U"
      "radial": [ 1.0, 2.0, 3.0, 4.0, 5.0, 6.0 ]
    },
    {
      "model": "Brown-Conrady U-D"
      "radial": [ -1.0, -2.0, -3.0, -4.0, -5.0, -6.0 ]
    }
  ]
  "undistortionOverscan": 1.2
}
```

> it should be:

```python
"lens": {
  "distortion": [
    {
      "model": "Brown-Conrady D-U"
      "radial": [ 1.0, 2.0, 3.0, 4.0, 5.0, 6.0 ],
      "overscan": 1.2
    },
    {
      "model": "Brown-Conrady U-D"
      "radial": [ -1.0, -2.0, -3.0, -4.0, -5.0, -6.0 ],
      "overscan": 1.1
    }
  ]
}
```

> Would you agree?

I (@JGoldstone) agree with Hendrik on this one, and this is the time to change it, before 1.0.0.

I'll stash what would be needed for a PR to do this, so we could rapidly review and merge if people agree it's the right thing to be done.

JGoldstone, 2025-02-10:
I could certainly amend the PR (or probably and preferably submit a new PR against main once the big PR is merged) to do as you suggest, but: what happened to the radial distortion coefficients? Are those still down in a dynamic/regular lens object, or did you just omit those in your last revision above?

h-fehlis, 2025-02-10:
The radial (and possibly tangential) distortion coefficients are still in the dynamic/regular lens object, just like the overscan values for that particular tracking sample. Only the "maxOverscan" values would (so far) go into the static lens object. Depending on how overscan is implemented on the graphics side, either the overscan values for each tracking sample (in the dynamic object) are used or only the "maxOverscan" values in the static lens object (if, for example, an image with a larger pixel resolution is rendered, which will have to be set once and cannot change with each new frame). By specifying both types of overscan values (current and maximum), we are covering both possible types of implementations on the graphics side.

JGoldstone, 2025-02-10:
OK. Right now, we have this:
```python
class StaticLens(CompatibleBaseModel):
    distortion_overscan_max: Annotated[UnityOrGreaterFloat | None,
      Field(alias="distortionOverscanMax",
            json_schema_extra={"clip_property": "lens_distortion_overscan_max",
                               "constraints": REAL_AT_LEAST_UNITY})] = None
    """Static maximum overscan factor on lens distortion. This is primarily
    relevant when storing overscan values, not in transmission as the
    overscan should be calculated by the consumer.
    """

    undistortion_overscan_max: Annotated[UnityOrGreaterFloat | None,
      Field(alias="undistortionOverscanMax",
            json_schema_extra={"clip_property": "lens_undistortion_overscan_max",
                               "constraints": REAL_AT_LEAST_UNITY})] = None
    """Static maximum overscan factor on lens undistortion. This is primarily
    relevant when storing overscan values, not in transmission as the
    overscan should be calculated by the consumer.
    """
```

I would propose that we define a new object holding per-model overscan max data that is comparable to this object used for dynamic distortion data:
```python
class Distortion(CompatibleBaseModel):
    # TODO ask the group about strictness of type checking.
    #    Let Pydantic convert silently?
    #    Restore strict=True?
    #    Change type to typing.Sequence? collections.abc.Sequence or collections.abc.MutableSequence?
    # radial: Annotated[tuple[float, ...], Field(strict=True)]
    # tangential: Annotated[tuple[float, ...] | None, Field(strict=True)] = None
    radial: Annotated[tuple[float, ...], Field(min_length=1)]
    tangential: Annotated[tuple[float, ...] | None, Field(min_length=1)] = None
    model: NonBlankUTF8String | None = None
```
and just as that object is accessible directly from the `Clip` via a `lens_distortions` accessor, there would be a comparable accessor for this unchanging data that includes overscan max.

But I'd suggest it be named something that allows us to add more unchanging, per-model distortion-related fields later.

I just am having a hard time naming that new class, and that new accessor, at the moment. And would love suggestions.

BTW I am convinced that this can readily be done, and moreover, that it should be done on top of the merge that puts us on Pydantic foundations — I don't want to block that for this.

h-fehlis, 2025-02-10:
Maybe it would make sense to row back a bit on what I said about `"distortionOverscanMax"` and `"undistortionOverscanMax"`.

Let's assume that in the future there is a third lens model supported by OpenTrackIO which describes how to distort points. Let's call it `"NewModel"`. Then we have `"Brown-Conrady U-D"` and `"NewModel"` both describing the same operation but with different mathematical methods (on the other hand, `"Brown-Conrady D-U"` describes the opposite "undistortion" operation).

If we define `"distortionOverscanMax"` to be the maximum overscan value that is required to handle distorting points **with either model** (`"Brown-Conrady U-D"` or `"NewModel"`), then we would indeed only need one `"distortionOverscanMax"` value and we would not need to differentiate between models. The same would hold true for `"Brown-Conrady D-U"` and all future additional undistortion models and their corresponding `"undistortionOverscanMax"` parameter.

We would lose the ability to add some "fine grain" per-model adjustments to the overscan maximum value but I assume that would likely not be a visible problem in practice. If an overscan value is chosen a bit larger than necessary, then the image quality of the distorted/undistorted image (depending on the direction of the operation) might suffer a small bit but probably hardly noticeable.

So to sum this up, if we define `"(un)distortionOverscanMax"` accordingly, we could leave the parameter names as they are now. However, I would still like to propose to move the (non-maximum) "dynamic" overscan values into the dynamic "distortion" object because these values will effectively only be used for best quality renderings (otherwise the graphics renderer might just use the maximum values).

I'd be interested in getting @jamesmosys 's opinion on this and also on my question in issue #163 as to why overscan values are _"primarily relevant when storing overscan values, not in transmission as the overscan should be calculated by the consumer"_ (see documentation text above) which seems to contradict my understanding.

closed 2025-02-18 by jamesmosys