SMPTE/ris-osvp-metadata-camdkit #161: SynchronizationPTP's `time_source` parameter should be named `leader_time_source`

JGoldstone, 2025-02-08:
If one looks at the `SynchronizationPTP` class now, three of the attributes that pertain to the PTP leader (formerly PTP master) have names starting with `leader_`:
`leader_identity`
`leader_priorities`
`leader_accuracy`
and one does not:
`time_source`

The other fields of `SynchronizationPTP` don't pertain strictly to the leader, and therefore don't have a `leader_` prefix:
`profile`
`domain`
`mean_path_delay`
`vlan`

To me, the time source belongs much more in the first group than the last.

jamesmosys, 2025-02-11:
Seems okay to me, but this needs comment from @makkbe

JGoldstone, 2025-02-11:
Yes, I've had an outstanding request for comment on that one by @makkbe for 12 days. Will nudge.

closed 2025-02-14 by jamesmosys