SMPTE/ris-osvp-metadata-camdkit #162: Source code of Fletcher checksum algorithm?

h-fehlis, 2025-02-10:
The checksum in the optional OpenTrackIO header (when not using ST2110) is calculated by using a simple form of the Fletcher16 algorithm (modulo 256) as described in RFC 1146 ("TCP Alternate Checksum Options", not an IETF standard), Appendix I (confusingly called "8-bit Fletcher Checksum" even though Wikipedia terms it "Fletcher16").

I could not find the short example source code for calculating a Fletcher16 checksum on https://www.opentrackio.org . Is it somewhere else in the documentation yet? AFAIK, we agreed on this version (on December 13th):

```
uint16_t fletcher (const uint8_t* data, uint16_t len)
{
  if (!data) return 0;
  uint8_t sum1 = 0;
  uint8_t sum2 = 0;
  while (len-- > 0) {
    sum1 += *data++;
    sum2 += sum1;
  }
  return ((uint16_t) sum1 << 8) | (uint16_t) sum2;
}
```

jamesmosys, 2025-02-11:
@makkbe please can you confirm this is the algorithm you implemented in python? If so we can add this to the template.html and link the github lines where it is calculated.

makkbe, 2025-02-11:
I asked ChatGPT for a C++ equivalent of my Python implementation, and here was the result:

```
uint16_t fletcher16(const uint8_t* data, uint16_t len)
{
    if (!data) return 0;
    uint8_t sum1 = 0;
    uint8_t sum2 = 0;
    while (len-- > 0) {
        sum1 = (sum1 + *data++) % 255;
        sum2 = (sum2 + sum1) % 255;
    }
    return (uint16_t(sum2) << 8) | uint16_t(sum1); 
}

```

jamesmosys, 2025-02-12:
Hmm this IS different - I think AI is hallucinating the modulus 255. @h-fehlis since sum1 is a uint8_t this is not required right?

h-fehlis, 2025-02-12:
Indeed. The modulo 255 operation can be found on Wikipedia, for example, but we decided to do a "modulo 256" which, as James noted correctly, is a no-op because storing the sum in an 8-bit integer automatically ensures only values between 0 and 255 are stored. If my memory serves me right, the modulo 255 operation has a larger statistical probability of detecting transmission errors but is also much slower and our emphasis was on speed and simplicity, so we chose the variant that was also once suggested to be used for alternative TCP checksums.

makkbe, 2025-02-12:
I don't have a C++ toolchain setup, but happy to run a few example payloads into my Python version if someone can do the equivalent on the C++ side, just for comparison?

PR #167 (2025-02-14): James/pre release fixes
Closes #154
Closes #162
Closes #163
closed 2025-02-14 by jamesmosys