SMPTE/ris-osvp-metadata-camdkit #163: "distortionIsProjection" boolean flag

h-fehlis, 2025-02-10:
The current documentation says about `"static" -> "lens" -> "distortionIsProjection"` that it is an _"Indicator that the OpenLensIO distortion model is the Projection Characterization, not the Field-Of-View Characterization. This is primarily relevant when storing overscan values, not in transmission as the overscan should be calculated by the consumer."_

I'm confused. I thought that the reason we are sending overscan values (from tracking systems) is to relieve the consumer (graphics systems) of the burden of calculating or estimating proper overscan values. Only the tracking system will usually know maximum distortion values for the entire zoom/focus range, so especially the maximum overscan values cannot be known by the graphics system beforehand.

In addition, I understand the difference between the "Projection Characterization" and the "Field-Of-View Characterization" but I learned that on the sender side (tracking system) we should not make assumptions about the receiver side (graphics) as to whether it uses an explicit shift in the projection matrix or not. Therefore, I believe we should always transmit overscan values that would work with both types of handling perspective shift (or projection offset?) even if they might be a little bit bigger than absolutely necessary. This would mean that the `"distortionIsProjection"` boolean flag would not be needed, right?

jamesmosys, 2025-02-11:
I agree the comment is not helping here!

My understanding is that - because of the difference in centre offsets, the required minimum overscan - or 'ideal overscan' will be different. There are multiple possible use cases - maybe we need to rethink how these are handled:

1. The tracking system or lens sends the ideal overscan every frame - calculated from the extremes of the image as described in OpenLensIO docs
2. No overscan is specified and the ideal overscan is calculated each frame by the render engine
3. The tracking system or lens sends an overscan value that is the maximum for any focus distance and or focal length of the current lens (this does not need to be a dynamic value)
4. No overscan is specified and the user manually enters an overscan to cover all cases in the render engine

I think our current parameters do cover these cases (with the changes proposed in #158), but the comments need to be improved and the language aligned with that in the OpenLensIO docs. (I like the term 'ideal overscan' for example.)

I think the distortionIsProjection flag is still required to deal with cases 1&2. But the comment is misleading.

h-fehlis, 2025-02-11:
My understanding so far was that we do cases 1 **and** 3 (send overscan and maxOverscan values). Cases 2 and 4 might nevertheless be done by the graphics system as a sort of "override" if the values that the tracking system sends should turn out to be inaccurate or if the tracking system simply does not send overscan values at all. But I believe these cases would be beyond the scope of OpenTrackIO.

In addition, calculating the right overscan values might not be that trivial (especially when mustache distortion occurs, a mix of barrel and pincushion distortion). Therefore, people on the graphics side might enjoy not having to do those possibly lengthy calculations for each frame when they could be done by the tracking system (where they could possibly be calculated and stored beforehand).

So if the tracking system does the calculation of overscan values but does not know if the graphics system compensates for projection offset using the projection matrix or a shader, I think it should not assume one rendering method and ignore the other. In some cases, even both rendering methods might be used, one for a real-time visualization and another in post processing. Therefore, I don't see the use for the "distortionIsProjection" flag.

My suggestion so far was to send an overscan (and maximum overscan) value that could be used for both rendering methods (the larger of both overscan values). Alternatively, I could imagine sending two overscan values (`"overscanMatrix"` and `"overscanShader"`? Or `"overscanFrustum"` and `"overscanFieldOfView"`?) and also send two maximum overscan values but I'm not sure if we want to make things even more complicated. To be clear, I have no problem with sending an overscan value for each rendering method, but I also accept if others find this too much trouble for, IMHO, relatively little gain.

h-fehlis, 2025-02-13:
Thanks, that's great! The same comment (about using maximum of values for both rendering methods) applies to the overscan values for each lens model in the (dynamic) lens distortion object as well, right?

PR #167 (2025-02-14): James/pre release fixes
Closes #154
Closes #162
Closes #163
closed 2025-02-14 by jamesmosys