SMPTE/ris-osvp-metadata-camdkit #183: Description of transforms list maybe misleading?

h-fehlis, 2025-04-04:
In the OpenTrackIO documentation it says about the "transforms" list:
_"A list of transforms. Transforms are composed in order with the last in the list representing the X,Y,Z in meters of camera sensor relative to stage origin."_
I believe this is only right if all transformations but the last have zero translation components (or are missing translation components entirely). I think it should be enough to just say:
_"A list of transforms. Transforms are composed in order."_
Or we could add that the X,Y,Z position in meters of the final transform that was composed of the list of transforms contains the position of the camera sensor relative to stage origin. Or am I missing something?

JGoldstone, 2025-04-14:
If you are going to say "in order" you would want wording that said what that order was, i.e. "Transforms are composed in sequential order, starting with the first transform in the list and concluding with the last transform in the list."  Or maybe I have the order backwards, given the statement about position relative to stage origin? Whichever it is, give me some wording and I can submit a PR to change it.

h-fehlis, 2025-04-16:
Your suggestion
_"Transforms are composed in sequential order, starting with the first transform in the list and concluding with the last transform in the list."_
seems fine to me. I would add something like:
_"The compound (or composed?) transform contains the position and orientation of the camera sensor relative to stage origin."_
It might be good to describe it mathematically, too, since this is not part of the OpenLensIO document. But that would be too much for the overview of parameters table. I'm thinking we might create such a document separately but the above explanation should be enough for most people.
I would be great if you could submit a pull request, thank you!!!

jamesmosys, 2025-06-23:
I am going to do a few doc updates shortly - I can include this

commit 7743b287 (2025-07-03): Documentation tweaks and re-adding dropFrame (#192)

* Add endianness to docs. Closes #181

* Clarification in transform docs. Closes #183

* Add segment note to docs. Closes #185

* Re-add dropFrame as an optional field in timecode. Closes #188

---------
closed 2025-07-03 by jamesmosys