SMPTE/ris-osvp-metadata-camdkit #185: Clarification of how to handle segmented OpenTrackIO packets.

jdw-epicgames, 2025-04-11:
The datagram for OpenTrackIO packets has a place holder for segmenting payloads over multiple transmission units.  However, there is no details on how to handle the need for retransmission or lost packets over a network.  If a payload is incomplete because of "transmission loss" how does one communicate back to the sender that a re-transmit is needed of a particular segment.   

Or, in this case, do we discard the entire payload (sequence) if we detect any lost packets?

jamesmosys, 2025-04-24:
Yes the latter - there is no retransmission of live data in the specification (or means to request it). Cross fingers that the next packet makes it through! In the case of a dropped frame in recorded data the missing intermediate keyframe would have to be interpolated from its neighbours.

I would propose adding the following for clarification:
'If a multi-segment payload is incomplete due to one or more segments being lost in transmission, the entire payload should be discarded.'
Would that suffice?

jdw-epicgames, 2025-04-24:
Yes that was our planned implementation. Thank you for the clarification.

commit 7743b287 (2025-07-03): Documentation tweaks and re-adding dropFrame (#192)

* Add endianness to docs. Closes #181

* Clarification in transform docs. Closes #183

* Add segment note to docs. Closes #185

* Re-add dropFrame as an optional field in timecode. Closes #188

---------
closed 2025-07-03 by jamesmosys